rommulaner/Anycubic_Vyper_MB_CE_6.2 #2: [BUG] After printing is finished, the table is pushed into the Y-axis belt tensioner.

pasha4ur, 2022-05-22:
### Did you test the latest `extui` code or prerelease?

No, I cannot complete

### Bug Description

Hi

Stock anycubic vyper.

After printing is finished, the table is pushed into the Y-axis belt tensioner.

I can hear the rumble of the belt slipping on the axle. On the stock firmware everything is ok. I'm printing with Octoprint. Slicer - Kura 4.

Y-axis belt tensioner https://www.thingiverse.com/thing:5230176

### Version of CR6Community Firmware

6.1d

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Mods

none

rommulaner, 2022-05-22:
Hi!

When you install a self design tensioner it is not a stock printer anymore. With that part you limit your travel in Y direction. It is not a Bug at all. If you want, you can change the travel limits in the configuration.h and compile a fitting version for yourself but it doesn't make sense to limit it for all other users if we implement it in general.

closed 2022-05-22 by rommulaner

pasha4ur, 2022-05-22:
I wrote that with Vyper's stock firmware all is ok.

On 6.1 was ok too.

pasha4ur, 2022-05-22:
The base mount of this custom tensioner shorter then stock on 5 mm. I checked.