rommulaner/Anycubic_Vyper_MB_CE_6.2 #3: [BUG] Cant save Linear Advance K-Factor

Dominik-1980, 2022-06-18:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

If the printer is turned off and on again, the K-Factor is set to 0.00
If I change it with M900 K0.45, save it with M500 and turn the printer off, next time it is back to 0.00

### Version of CR6Community Firmware

CE_6.1d

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

rommulaner, 2022-07-04:
I tried changing the value and it was saved correctly. Is it perhaps written in the Start gcode or in the print gcode so that it will be saved when you tip on screen?

rommulaner, 2023-11-05:
As there is no more feedback, this issue is closed for now. 
If this is still a problem, let me know.

closed 2023-11-05 by rommulaner