rommulaner/Anycubic_Vyper_MB_CE_6.2 #6: [BUG] Newer versions (motherboard v0.0.6) temperature reading 20c low.

fauks, 2022-07-09:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

After flashing 6.1d, the nozzle temperature read ~20c lower than it really is. I changed thermistors to make sure it wasn't a bad thermistor. After flashing back to the stock firmware, the problem goes away.  

The stock firmware that corrected the issue is located here:
https://drive.google.com/file/d/1hkOLsbbNljoXVjasLwcvBRqEbkb5RCyu/view

### Bug Timeline

new

### Expected behavior

Read the correct temperature within reason. 

### Actual behavior

Reading significantly low, estimated 20c low. TPU would print horrible at 220c and great at 180c. After flashing back to stock, it prints great from 200-220c. 

### Steps to Reproduce

Instal 6.1d on any newer Vyper with the changed hotend and v0.0.6 motherboard. 

### Version of CR6Community Firmware

v0.0.6 motherboard. MB 6.1d firmware, v3.5 (recalibrated) screen firmware

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

ZerGo0, 2022-09-11:
Hey, have your tried the latest official version v2.4.5 which can be found on the anycubic site?

rsmith985, 2022-09-12:
So I just had to get my motherboards and hotend replaced due to issues.  My hot end is what appears to be their "new" style and is not like the original that came with my printer a year ago.

After installing the latest firmware here.  I agree that something seems off with the temperature.  I mostly only print in PETG and it seems like I need to print in a much lower temperature than before.  I printed several temp towers, and I agree that it seems to be off by roughly 20deg.  The same roll I printed prior to things breaking had a sweet spot around 240, and now the sweet spot seems closer to 220.  I was also able to print PETG at temps less than 200 which was not possible previously.

I have not tested with stock firmware yet to see if the problem goes away.  And I haven't tested with a different type of filament like TPU.

Even though I don't like that I can't trust the temperature anymore, for the moment I think I will just subtract 20deg from all my old settings, since I really don't want to go back to using the stock firmware.

ZerGo0, 2022-09-12:
I'm pretty new to 3d printing and Marlin, but I compared the public version of Anycubic's Vyper FW Configuartion.h with the one used here and I have noticed that there are some differences related to thermals. 

https://www.diffchecker.com/VciVHsL9

Starting at line 497 (left file), the default pid values seem to be changed, but I'm pretty sure that there is a auto calibration in the menu anyway? Not sure if that could be it, I don't have my printer yet, just trying to help.

aspiringnobody, 2023-09-13:
I can confirm that this isn't just one instance.  I'm also getting abnormally low E0 temps, but only with your version of the CF.

Is there a way to hard-code an offset or something so I don't have to manually change my slicer back and forth between 180C for my Vyper and 200C for all my other printers?

aspiringnobody, 2023-09-13:
> > So I just had to get my motherboards and hotend replaced due to issues. My hot end is what appears to be their "new" style and is not like the original that came with my printer a year ago.
> > After installing the latest firmware here. I agree that something seems off with the temperature. I mostly only print in PETG and it seems like I need to print in a much lower temperature than before. I printed several temp towers, and I agree that it seems to be off by roughly 20deg. The same roll I printed prior to things breaking had a sweet spot around 240, and now the sweet spot seems closer to 220. I was also able to print PETG at temps less than 200 which was not possible previously.
> > I have not tested with stock firmware yet to see if the problem goes away. And I haven't tested with a different type of filament like TPU.
> > Even though I don't like that I can't trust the temperature anymore, for the moment I think I will just subtract 20deg from all my old settings, since I really don't want to go back to using the stock firmware.
> 
> So I just ended up using a new thermistor and changed and recompiled the firmware to use this new thermistor. Look on amazon for the bullet type thermistor designed for the Artillery Genius printers.
> 
> Then download VS code and the source files here on github. Look up how to compile Marlin, its very easy.
> 
> In Configuration.h, there is a thermistor section you can change.
> 
> If you end up going this route, let me know and I can rebuilt it using that thermistor if you aren't comfortable doing it yourself.

I'd appreciate a binary blob if you don't mind.  I'd rather not have to figure out VS Code for the Vyper.  I use iVision to compile Marlin for my other Anycubic so I'm not familiar with using VS Code for it.  Also a link for the thermistor would be helpful.  The ones that come up when I search for the Artillery Genius seem to be the same as what I already have.

Thanks,
Evan

rommulaner, 2023-09-18:
Hi!

In general it’s no problem to adapt the temperature to your belongings. I
just would change the map for the Thermistor a bit and then it’s fine. I
would compile you a version of that, that’s no problem.

Greets
Robert

rommulaner, 2023-09-18:
New FW as BIN in zip file compiled with changed temps.

closed 2023-09-18 by rommulaner