rommulaner/Anycubic_Vyper_MB_CE_6.2 #7: [BUG] Printer stopped calibrating bed after switching off during calibration

user897943, 2022-08-10:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

I started bed calibration at 85C temp on the bed, then got a shut down warning error, power cycled the printer and it refused to level, it would home, then the nozzle would just sit there and do nothing. Factory resetting to defaults did nothing. Then I reinstalled the 2.4.5 firmware from Anycubic (factory firmware) and it's back to normal, so it looks like some parameters got corrupted on your version and restoring to factory did not really restore/zero values.

### Bug Timeline

Started using your fw a week ago, and noticed it today

### Expected behavior

I expect factory defaults to zero the values to clear any eeprom corruption

### Actual behavior

Did not appear to clear values, as the bed leveling was still not working, although reinstalling anycubic original firmware restored levelling so not a hardware issue.

### Steps to Reproduce

See above

### Version of CR6Community Firmware

6.1d

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

IdeaMaker

### Host Software

OctoPrint

rommulaner, 2022-11-29:
Not really a bug when power off during leveling.

closed 2022-11-29 by rommulaner

user897943, 2022-11-29:
You didn't understand. The printer stopped calibrating PERMANENTLY. I had to put the stock firmware in the EEPROM to get the printer to do bed levelling again. Then I reinstalled the community firmware and it was back to normal. So a factory reset is not clearing all the values after all... Anyway I no longer use this firmware, use Klipper instead so if you wanna close and leave the bug in the firmware it won't affect me but at least other users know how to recover the function.

rommulaner, 2022-11-29:
I powered off several times during leveling and it always ran perfect after that. In your case the Firmware got damaged during you powered off during writing in my opinion. 
So it's not a Bug.