rommulaner/Anycubic_Vyper_MB_CE_6.2 #8: [BUG] Setup Menu not opening

ZerGo0, 2022-09-15:
Whenever I flash the MB Firmware and try to get into the Setup Menu the entire touchscreen stops working. All the other menu opens work just fine, I can the Temps as well.

How I flashed it:
1. Flash LCD
2. Touch all 4 corners (I don't think that I have to do that, but I do it anyway)
3. Flash MB (I have to do this multiple times, because I'm sometimes stuck in the loading screen after the jingle or the screen is just blank)
4. Test the touch screen
5. Click the setup menu

After rebooting I'm not able to get passed the loading screen, I'm always stuck at like 90%. I'm not sure if that is because I didn't do a factory reset or not?

rommulaner, 2022-09-15:
What Viper version do you have?
Try to use computer to reset EEPROM.

ZerGo0, 2022-09-15:
It's a v0.0.6 MB, not sure how to check the screen version. I couldn't even get into the main menu after clearing the EEPROM.

ZerGo0, 2022-09-15:
Got it working, I built bot firmwares myself and apparently the LCD one is kind of broken right now? Not sure what exactly, but using the one in the zip worked just fine.

closed 2022-09-15 by ZerGo0