rommulaner/Anycubic_Vyper_MB_CE_6.2 #9: [BUG] CFW 6.1e AnycubicVyper

auratum, 2023-03-19:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

Automated bedleveling isn´t working.
Every Time i try to level the bed, the nozzle eather touches the first point an goes back to home again, telling me to set up Z-offset, but even if i do so, it´s the same again or the extruderhead drives to the middle of the bed and than upwards, without leveling anything.
I can not find a solution for this issue.

### Bug Timeline

Occured after flashing from 6.1d to 6.1e

### Version of CR6Community Firmware

6.1e

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

rommulaner, 2023-03-26:
There is no Version 6.1e from me. I only have the 6.1d.

rommulaner, 2023-05-02:
> I have the same issue with 6.1e. I upgraded from the official 2.4.5.

Hi!
6.1e is not my version, I did only 6.1d.
Greets
Robert

auratum, 2023-05-02:
> I have the same issue with 6.1e. I upgraded from the official 2.4.5.

Like he said, it´s not his FW. Don´t use the Version e, use Version d. That´s stable, you know the source and can get Support.
At this time, the suspicious Version e causes a lot of trouble and bricks the printers from time to time.

pixelkorrigierer, 2023-05-02:
Thanks for this information. Good to know that this is a buggy version. 
I will try then your 6.1d. Be aware if that does not work! ;-)

I came across the 6.1e from [Pasha4ur's blog](https://pasha4ur.org.ua/articles/anycubic-vyper-overviews-guides-adjustments-settings-tips-modifications-upgrades-and-custom-firmwares/8?fbclid=IwAR11sZy93dq9X_i9YA7ea2ryaQ5gn1hltEU44LHKiP98ZpvV7HiY5pbWY4U). This version has nice features. Well, does not help if it does not work.

pixelkorrigierer, 2023-05-02:
Changed firmware to your 6.1d. Works like a charm so far. Thanks a lot for your support!

closed 2023-11-05 by rommulaner