rommulaner/Anycubic_Vyper_MB_CE_6.2 #10: How to upgrade this to latest Marlin release V2.1.x?

BassXX, 2023-04-04:
Hello,

is it possible to upgrade this firmware to the newest marlin version?

I would like to use input shaping and was wondering how to upgrade?

Can you just compile this with the newest marlin files? Or is there more to take into consideration?

Thanks and regards

rommulaner, 2023-04-08:
It's not that easy as you think. You need to use the config from the new version and rework it completely. And you also need to rework all other changes that were made in any file of Marlin. This will take some time...

BassXX, 2023-04-11:
Hmm, too bad. Unfortunately, I think I am not familiar enough with Marlin Firmware to do so without screwing something up

rommulaner, 2023-12-23:
Solved with new Release 6.2 based on Marlin 2.1.2.1.

closed 2023-12-23 by rommulaner