rommulaner/Anycubic_Vyper_MB_CE_6.2 #11: [BUG] Unable to compile

thomas41587, 2023-07-09:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

I am unable to compile the files. I already cleaned the C:\Users\<user>\.platformio\packages\framework-arduinoststm32 fodler to make sure that this does not cause the issue. Here is the output of the compiler:

```
Processing STM32F103RET6_ac_tri_f1 (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.10900.200819 (1.9.0)
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- IWatchdog @ 1.0.0
|-- Servo @ 1.1.2
|-- Wire @ 1.0
Building in release mode
RuntimeError: deque mutated during iteration:
  File "C:\Users\Thomas\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 182:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "C:\Users\Thomas\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Thomas\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Thomas\Desktop\Anycubic_Vyper_MB_CE_6.1d\buildroot\share\PlatformIO\scripts\offset_and_rename.py", line 24:
    marlin.relocate_vtab(LD_FLASH_OFFSET)
  File "C:\Users\Thomas\Desktop\Anycubic_Vyper_MB_CE_6.1d\buildroot\share\PlatformIO\scripts\marlin.py", line 33:
    replace_define("VECT_TAB_OFFSET", address)
  File "C:\Users\Thomas\Desktop\Anycubic_Vyper_MB_CE_6.1d\buildroot\share\PlatformIO\scripts\marlin.py", line 22:
    for define in env['CPPDEFINES']:
================================================================================================== [FAILED] Took 4.86 seconds ==================================================================================================
Environment              Status    Duration
-----------------------  --------  ------------
STM32F103RET6_ac_tri_f1  FAILED    00:00:04.861
============================================================================================ 1 failed, 0 succeeded in 00:00:04.861 ============================================================================================ 

 *  The terminal process "C:\Users\Thomas\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
```

### Version of CR6Community Firmware

Anycubic_Vyper_MB_CE_6.1d

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

rommulaner, 2023-08-04:
Did you modify some configuration files or why do you want to compile?

rommulaner, 2023-08-04:
A compiled file can be found in the zip file.

rommulaner, 2023-08-04:
Same for the display.

spazzb0t, 2023-10-17:
I'm upgrading thermistor for my dragonfly hot end and want to raise the max print temp to 300°C. The only way I am aware of is by editing the configuration and compile, but I cant get around the error described. Can you help me out?

rommulaner, 2023-10-17:
Sure, no problem. 
Send me the configuration files at rommulaner@gmail.com and I compile it for you.

rommulaner, 2023-11-05:
An updated version of this branch will be uploaded now, then compiling will be possible with the current PlatformIO.

closed 2023-11-05 by rommulaner