rommulaner/Anycubic_Vyper_MB_CE_6.2 #12: Auto leveling > Printer powercycle

FraatGIT, 2023-12-24:
When I'm trying to run auto leveling the printer is power cycling. Sam issue with 6.1e and 6.2. Tried with factory reset don't helped.

rommulaner, 2023-12-24:
Updated the current release of 6.2 to leveling settings of 6.1d.
Always level without bowden and already start printer without the bowden.

FraatGIT, 2023-12-25:
thanks for the fast response, now the auto leveling will stop after the 4th measurement point. Under setup info I see 6.1d (la+is). In the main screen i see Anycubic Vyper CE-6.2 ready.

rommulaner, 2023-12-25:
Thanks for the hint with the version number. I missed that.
The leveling stops because of the bowden tube that triggers the sensor. 
You need to remove the bowden tube BEFORE you start the printer, because ther the sensor gets zeroed, then everything works fine.

closed 2024-01-18 by rommulaner