rommulaner/Anycubic_Vyper_MB_CE_6.2 #14: Autolevelig

roubal82, 2024-01-03:
Good day,
autoleveling ends after two steps.
It does this even after earning a bowden.
Leveling works in firmware version 6.1b.
Please advise, thank you

rommulaner, 2024-01-03:
Have you used the actual Release update from 27.12.2023? Before there was a bug in Sensor Tare.

roubal82, 2024-01-03:
Yes from 12/27, which file is for the unmodified motherboard?

rommulaner, 2024-01-03:
The 4x4 or 7x7 (mesh size).
What is happening exactly? The first point is probed normally with a value in the end on the display and then?

roubal82, 2024-01-05:
it is for 4x4 and 7x7.
the first point is traversed to find the rough height, then traversed again for accuracy.
if it continues to point 2, it writes the height, otherwise it returns to the starting position.

roubal82, 2024-01-05:
The only improvement is the new firmware.
If the value is written, it is 2.125.
Which file to use? La_is or E0_300C?
The motherboard is not modified.

rommulaner, 2024-01-05:
You should use the LA_IS. The E0 300C is the same but allows 300 degrees on the Hotend if it is upgraded. 
Did you use 6.1b or 6.1d before?

Over 2mm is very high, I have nearly 0 at leveling. I think that's the problem. At Sunday I can try to modify the firmware for you and you can try that version.

roubal82, 2024-01-05:
I use version 6.1b, and it shows values from 3.6 to 4.1 when leveling.
I haven't figured out a way to reset the values to start at 0.

rommulaner, 2024-01-07:
[main_board_LA_IS_4x4_high_nozzle.zip](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/13855024/main_board_LA_IS_4x4_high_nozzle.zip)
Please test this version with higher nozzle if that works for you and give me feedback what values are now displayed at leveling.

closed 2024-01-18 by rommulaner

roubal82, 2024-01-18:
thanks for the edit.
is that file even at a temperature of 300 degrees

roubal82, 2024-01-18:
I'll test it today and let you know

rommulaner, 2024-01-18:
I have compiled two 300°C MB BINs for you. They are in the release.

roubal82, 2024-01-18:
I uploaded "hight nozzle" but it didn't even write the first value.
Fro example of what version 6.1b writes:
4.548 - 4.248 - 3.977 - 3.849
4.444 - 3.886 - 3.825 - 3.651
4.263 - 3.809 - 3.581 - 3.556
4.031 - 3.733 - 3.473 - 3.309

rommulaner, 2024-01-22:
Is now adjustable with 6.2.1 (Pre-Release). Please flash MB and LCD.
In the LCD Release infos is described how to do a bed leveling with the new feature.
Please test and give feedback.