rommulaner/Anycubic_Vyper_MB_CE_6.2 #15: Screen Freezes

mgochis, 2024-01-09:
Printer works great when controlled with Octoprint but screen freezes after a few touches or from just being on for a while. I've tried several times to reinstall the firmware on both the screen and board. It doesn't happen at any specific menu or time and appears to be random. The screen stays on and freezes on whatever menu was up at the time. Touch panel is totally non responsive once it freezes. Only way to clear it is to restart the printer.

rommulaner, 2024-01-09:
Did you use the current releases? They were updated end of December.

I also had such a behavior from time to time, no response of touch screen (no beep) followed by an automatic reboot of the printer, but only when there was an SD with gcode in the slot. I have reset to default settings two times in a row and formatted the SD. So far this error has not occurred again.

Please flash the current available resleases and reset to default settings two times (hold for >2sec). 
You should also reset the display by holding down the Anycubic text for >2sec to get to the boot screen and hold on the top of the screen for >2sec. By clicking on the boot logo you will get back to the main screen.

rommulaner, 2024-01-18:
Seems to work, because of no more feedback.

closed 2024-01-18 by rommulaner

rommulaner, 2024-12-11:
Is there something connected with printer via USB?

nilsbyte, 2024-12-11:
> Is there something connected with printer via USB?

No, I unplugged my Octopi Instance for the whole process and it is still unplugged.

rommulaner, 2024-12-11:
Then I think the flash was not fully successful. Try a reflash.
If that doesn't fix it, use the reset file for the screen and reset the screen and reflash it.

nilsbyte, 2024-12-11:
> Then I think the flash was not fully successful. Try a reflash. If that doesn't fix it, use the reset file for the screen and reset the screen and reflash it.

Okay I have run the reset (3 files in DWIN_SET folder), and immediately on the next power up, I reflashed the display. after END was displayed, the printer did beep: 1 short beep, 1 long beep. 

I did a power cycle and nothing, black display (only printer jingle). power cycling a few times and the display shows the menu, but same as before, when tapping anything, the display freezes.

here is a photo of the flash: 

[image]

rommulaner, 2024-12-11:
Ok. Why is the display on V4.0? It should be V3.5. Is the BIN T5L_UI_DGUS_V35_20200730.bin in the DWIN_SET folder?

nilsbyte, 2024-12-11:
> Ok. Why is the display on V4.0? It should be V3.5. Is the BIN T5L_UI_DGUS_V35_20200730.bin in the DWIN_SET folder?

Yes, it is.

[image]

nilsbyte, 2024-12-11:
> Maybe your Display OS is too new and don't accept the V35. You can try to add the attached additional BIN files to the folder, perhaps it works with that. But I don't know if it works and fits to your display. Try on your own risk.
> 
> [T5L_UI_DGUS2_V35_20200730_2.zip](https://github.com/user-attachments/files/18096844/T5L_UI_DGUS2_V35_20200730_2.zip)

Which of the BIN files should i copy on the SD card?

[image]

nilsbyte, 2024-12-11:
Had to put all 3 files on the SD card inside the `DWIN_SET` folder.

[image]

The red text shows the old firmware, after that, the firmware stayed 3.5. I reflashed the 6.2.2 display firmware and not it works! Thanks.

If the purpose of the `T5L_UI_DGUS2_V35_20200730.BIN` in the CE-6.2.2 firmware is to downgrade, then you should add the 2 other files from the zip.