rommulaner/Anycubic_Vyper_MB_CE_6.2 #16: Z-Axis higher after flash

trashmaster9000, 2024-01-11:
i've just flashed this fw to my vyper and printed a 20mm test cube.
z-Axis is around 10% higher than with the stock fw.
i checked the e-steps they seem correct (400/mm)

what can cause this issue?

rommulaner, 2024-01-11:
400 Steps/mm should be correct. Can you provide the gcode file you printed?

trashmaster9000, 2024-01-12:
of coarse
messured again, with the old FW this cube has a height of 19.9mm with the new custom FW it is 20.9mm
should be 20mm, both printed with the exact same file.

had to rename it to .txt for uploading...

[AV_Calibrate.gcode.txt](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/13921539/AV_Calibrate.gcode.txt)

rommulaner, 2024-01-12:
Ok, that's interesting...
You did an Autoleveling on both before the print?
How is the fade height identical in both situations?

trashmaster9000, 2024-01-12:
yes, i did on both a bed leveling

what is the fade height?

rommulaner, 2024-01-12:
That z-value in the Leveling Settings is for fading out the bed leveling adjustment during printing. So if there are 10mm set, the software reduces the bed mesh correction in each layer to get at Z10 zero correction. You can set it to 0 and check if this is the problem. 
But mainly it's then a Marlin related problem and has nothing to do with my changes.

trashmaster9000, 2024-01-12:
did a new bed calibration, these are the values.
is it correct that all values are negative?

i've changed fading to 1mm and currently printing again, will let you know once ita finished.

[image]

trashmaster9000, 2024-01-12:
ok, the print failed after 1mm. changed to 5mm and it failed after 5mm.

it looks like its skipping a few layers.

[image]

rommulaner, 2024-01-12:
The values of the mesh vary from printer to printer and the installation of the nozzle.
What values do have roughly in the bed mesh map? (min max)
Ok, that's strange. Seems to be a Bug in Marlin itself. You can set Fade Height to 0 to deactivate fading, then it should work.

trashmaster9000, 2024-01-12:
can you see the images i've attached?

i've set fade to 0 and now it seems like its working, i get the exact same cube as with the original FW.

rommulaner, 2024-01-12:
Now I can see the pictures.
I think there is a bug in Marlin that misses to calculate the middle plane of the leveling and fades out the whole values instead of the deviations. 

Another workaround would be to adjust the manual home position for z to get as close as possible to the bed surface at z=0 after homing. Then the values in the bedmesh will be around close to 0 and the fade out will work. You can do that deleting the leveling mesh, then home all axis, move the nozzle to center of the bed and adjust Z until it it touches the bed. This value then needs to be added to the existing value of 1.4. If you have a positive value at touching the bed, the manual home position of z needs to be lowered by your measured value. If it's negative, you need to raise the value by your measurement.

trashmaster9000, 2024-01-13:
should i report this issue to the marlin fw developers then?

i'm trying to get prusa slicer to work and wasted a lot of time troubleshooting since the test cubes all are higher again and i thought its a issue with prusa slicer, only to discover fading doesnt get saved/loaded after a restart.

i've discovered another bug:
fading distance is not saved despite the display displaying the message "settings stored"
if i power cycle the printer all settings (temperatures and fading distance) in the calibration settings panel are not properly restored.

if i click the "better accuracy" button the values are stored and get loaded correctly after a power cycle

rommulaner, 2024-01-13:
Yes you can report that issue to the marlin devs and see what they reply.

About the save settings issue, I think I found the a issue in the back button process at the display.
If you want you can test it with the flash files attached onto the display if you want and give me feedback. I would apprechiate that. I'm not at my printer now.

I've changed the back button to a requested save at the following pages:
- Change Filament
- PID Tuning
- E-Steps Calibration
- Tune Retraction
- Leveling Settings
- Mesh Validation

[Vyper_6.2_SAVE_SETTINGS_TEST.zip](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/13929486/Vyper_6.2_SAVE_SETTINGS_TEST.zip)

trashmaster9000, 2024-01-14:
i've tested your new FW, it looks like everything gets stored correctly now.
also fixed retraction settings not getting stored.

thank you for the quick help and fixes!

rommulaner, 2024-01-14:
Perfect! Thanks for testing and your feedback. I will upload this changes in the repository and update the release.

closed 2024-01-18 by rommulaner