rommulaner/Anycubic_Vyper_MB_CE_6.2 #18: E-steps calibrating stuck on preheat

Mootfrost777, 2024-01-31:
Hotend heats to 190 and starts heating up and down +- 3degrees but doing no calibration. reboot required. What

rommulaner, 2024-01-31:
Do a PID calibration before anything else. If the temperature cannot be hold stable, leveling, extruding and probing doesn't work as expected.

Mootfrost777, 2024-01-31:
> Прежде всего выполните калибровку ПИД-регулятора. Если температуру невозможно поддерживать стабильной, выравнивание, выдавливание и зондирование не работают должным образом.

Thank you sou much for the advice, esteppers passed. But bed leveling still starts and immediately ends without doing anything

rommulaner, 2024-01-31:
Perhaps the nozzle is too high and it runs to the set limit of -3mm and that's why it's aborting the leveling.
I had currently one with a similar problem and I will change that in a new built today. I will inform you when it's available.

rommulaner, 2024-01-31:
6.2.1-2 is now available for Mainboard please test with that version. Display has not changed.

Mootfrost777, 2024-01-31:
It works! Thank you for your help.

closed 2024-01-31 by Mootfrost777