rommulaner/Anycubic_Vyper_MB_CE_6.2 #19: Build fails unless input shaping is enabled

smcbride-ca, 2024-02-02:
When attempting to build the firmware with input shaping disabled, the build fails with the following errors:
```
Marlin/src/lcd/extui/dgus_creality/creality_touch/InputShapingHandler.cpp: In static member function 'static void InputShapingHandler::Init()':
Marlin/src/lcd/extui/dgus_creality/creality_touch/InputShapingHandler.cpp:23:26: error: 'class Stepper' has no member named 'get_shaping_frequency'
   23 |     set_freq_x = stepper.get_shaping_frequency(X_AXIS);
      |                          ^~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/extui/dgus_creality/creality_touch/InputShapingHandler.cpp:24:26: error: 'class Stepper' has no member named 'get_shaping_damping_ratio'
   24 |     set_zeta_x = stepper.get_shaping_damping_ratio(Y_AXIS);
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/extui/dgus_creality/creality_touch/InputShapingHandler.cpp:25:26: error: 'class Stepper' has no member named 'get_shaping_frequency'
   25 |     set_freq_y = stepper.get_shaping_frequency(Y_AXIS);
      |                          ^~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/extui/dgus_creality/creality_touch/InputShapingHandler.cpp:26:26: error: 'class Stepper' has no member named 'get_shaping_damping_ratio'
   26 |     set_zeta_y = stepper.get_shaping_damping_ratio(Y_AXIS);
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/extui/dgus_creality/creality_touch/InputShapingHandler.cpp: In static member function 'static void InputShapingHandler::HandleInputShapingBackButton(DGUS_VP_Variable&, void*)':
Marlin/src/lcd/extui/dgus_creality/creality_touch/InputShapingHandler.cpp:33:13: error: 'class Stepper' has no member named 'set_shaping_frequency'
   33 |     stepper.set_shaping_frequency(X_AXIS, set_freq_x);
      |             ^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio/build/trigorilla_v006/src/src/lcd/extui/dgus_creality/creality_touch/InputShapingHandler.cpp.o] Error 1
```
There are no issues building when input shaping is enabled. 
[Full build log](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/14145329/builderror.txt)
[marlin Configuration](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/14145304/Configuration.zip)

rommulaner, 2024-02-03:
Why do want to compile it that way? You can deactivate IS with frequency set to 0.

rommulaner, 2024-02-16:
Issue will be fixed in the coming update.
Compilation without Input Shaping is then possible but the button on the screen is still there with no function.

closed 2024-02-18 by rommulaner