rommulaner/Anycubic_Vyper_MB_CE_6.2 #20: Cant adjust the mesh-bed-level

FunkyOwl-Net, 2024-02-04:
Hi,

i sucsessfully installed the CFW on my Vyper without any issues. The only thing which is not working is the Z-Offset adjustment off the measurement points, i can access the menu and also write a number in it but it wont change. What am i doing wrong?

Thanks

FunkyOwl-Net, 2024-02-04:
Yes right, i can write my value with the keyboard and after i push the green arrow for confirming the Z-Offset for the measuremend point it wont do any changes, also after closing and reopening the menu, the old values are still there, no changes.

rommulaner, 2024-02-04:
This is only possible after running leveling is done successfully.
Maybe you need to flash both again and make firmware reset. 
For accurate measurements you should activate Better Accuracy to turn off the Hotend heater during probing.

FunkyOwl-Net, 2024-02-05:
Thanks for Info and also for your great work on the CFW. I will do the fresh install of the CFW again, when i'm home. How do i reset the Firmware? Maybe thats the issue because i had also some layer shifting which i didnt had bevore.

When i was doing the Auto-Leveling i already activated the better accuracy.

rommulaner, 2024-02-05:
In the Setup menue press and hold the reset button.
The layer shifts are because of the higher limits in the firmware, you need to find your own limits for your printer or check the driver voltages and belt tension.

FunkyOwl-Net, 2024-02-05:
I will try at home today evening, thanks. Is there any standard driver voltage which i can set as a value? I have the standard ones which came with the Vyper, i did not change the stepper motors, or is the only way to measure with a multimeter? Sorry for the dumb questions, im only 3 weeks into 3D printing now and thats new for me.

rommulaner, 2024-02-05:
You need to measure it with a multimeter. Here a good video: https://youtu.be/SN_-bNceWUI?si=EHpA1hBV0uMzr-8b
If you still have layer shifts you need to setup your slicer to values that work.

FunkyOwl-Net, 2024-02-08:
I installed the CFW 7x7 new and also resettet the firmware, after that i started the auto leveling which was successfull, but i cant adjust the values of the measured mesh. What can i do?

Layer shifting seems to be gone

rommulaner, 2024-02-08:
Does this also exist at the 4x4 version? There I know it works on my printer.
At the 7x7 it is not possible to adjust the hidden values, that's for sure. But it should be possible for the shown values.
Why do you want to adjust the values?

rommulaner, 2024-02-16:
I checked the code and it is only possible for 4x4 mesh to edit the values. For the 7x7 grid it is not possible as it makes no sense because you can't select the probed points between the displayed values. So it works as it should. If you want to have editable values, you need to use the 4x4 grid.

closed 2024-02-16 by rommulaner