rommulaner/Anycubic_Vyper_MB_CE_6.2 #21: component cooling fans make high-frequency noises

LeonardoHome, 2024-02-06:
Version 6.1b was previously installed on my printer. 

Now I have updated to the latest version 6.2.1-2. With this update, the component coolers now always make high-frequency noises when they are not running at 100%. 
I have tested all versions 6.2.X and it is the same on all versions. 

If I downgrade back to version 6.1b, the fans are nice and quiet again. 

I admire the new firmware, especially the 7x7 mesh, but I can no longer stand next to the printer with the fan noise.

rommulaner, 2024-02-06:
Ok, that's new to me. My Vyper is quite happy with that PWM frequency. 
I can change that to old value, no problem. I took the standard value that was already in preconfigured in Marlin. I compile the 7x7 Standard Version for you and you can test if that works with your Vyper.

LeonardoHome, 2024-02-07:
Thank you very very much, that would be great!

rommulaner, 2024-02-07:
Here the 7x7 standard version with the 100Hz FAST_PWM as the 6.1d has it.
Please test and give me feedback.
[main_board_LA_IS_7x7_PWM.zip](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/14201805/main_board_LA_IS_7x7_PWM.zip)

LeonardoHome, 2024-02-07:
I tested it, but unfortunately this version did not fix the problem.

rommulaner, 2024-02-07:
I don't have the source code of the 6.1b. I have compared the 6.1c and 6.1d and there was the same settings.
But the original firmware from Anycubic has FAST_PWM disabled and I took that into the attached firmware.
Can you test this one and give me feedback?
If that doesn't work I don't think there is a another way...
[main_board_LA_IS_7x7_PWM_2.zip](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/14202485/main_board_LA_IS_7x7_PWM_2.zip)

LeonardoHome, 2024-02-08:
I tested it, but again the same result. The fan still makes noises.
I flashed the 6.1d from you and the fans are silent again.

So is there no way to geht the fan control of the 6.1d into the 6.2.1-2 version?

rommulaner, 2024-02-08:
Ok, so it has something to do with a change from 6.1d to 6.2...
I investigate what could be cause.
Thank you again at this point.
I'll come back when I have found something...

rommulaner, 2024-02-15:
I'm sorry, but I don't find anything that can cause this problem. The FAST_PWM code is a bit different but the result should be the same. I'll try to make some frequency tests at the weekend, perhaps this works.

LeonardoHome, 2024-02-15:
Thank you very much for your investigation so far.

Is it possible to generate a 6.1d firmware with the 7x7 bed mesh? Because the 6.1d works fine so far, but i am missing the 7x7 bed mesh.

rommulaner, 2024-02-16:
I've now tested different settings and nothing worked. But then I switched to the old mainboard setup and the fans are now quiet.
Here the version 7x7 for testing: 
[main_board_LA_IS_7x7_PWM_3.zip](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/14314846/main_board_LA_IS_7x7_PWM_3.zip)

LeonardoHome, 2024-02-18:
I testet this version and it is much better than the "normal" 6.2.1-2. You can still hear the frequency of the fans but it is no longer a high-pitched sound, but more of a low-frequency hum. 
It not as quiet as with the 6.1d but much better than with the standard 6.2.1-2.
Thank you so far.

closed 2024-02-18 by rommulaner