rommulaner/Anycubic_Vyper_MB_CE_6.2 #22: Compilation (LA_TE_IS) errors

KAPTOHKA, 2024-02-13:
Hi, 
I can't compile *(LA_**TE**_IS), it gives yellow error...
*bin-file creates, 
but after the firmware the printer beeps and constantly reboots to the main screen...
*(LA_IS) compiles with the same yellow errors,
after the firmware the touchscreen works, goes through the menu,
but I can't check the work, because the board is redone to UART, need TE
Where should I dig to make it work?

[image]

[image]

KAPTOHKA, 2024-02-13:
> Did you only flash the Mainboard?

The screen and motherboard are flashed...
*TE* - doesn't work(((
Not *TE* - the screen and menu work)))
Can you compile  *(LA_TE_IS)* at home and post it here?

rommulaner, 2024-02-15:
I've gone through the TMC settings and aligned them to the settings from the 6.1 and I think it should work now.
The warnings has nothing to do with this issue but I added some lines of code to avoid them.
Can you please test the attached firmware and give me feedback if it works now?
[main_board_LA_TE_IS_7x7.zip](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/14301107/main_board_LA_TE_IS_7x7.zip)

rommulaner, 2024-02-16:
I've tested the TMC setting on my vyper without UART mod and my printer also restarts in a loop like yours. I tried to figure out what's the problem, but it seems that is a Marlin related issue that I can't fix it at the moment. I have no clue why it behaves like that. Related to the TMC only Marlin has changed but the settings are the same as in the 6.1d.
I tried it also with 6.1d and there the printer starts fine and the TMC menue appears.

KAPTOHKA, 2024-02-18:
* I did some digging in the code, the last time I did it was more than a year ago and I don't remember much (((
* You don't have configuration ```Configured.h``` and ```Configuration_adv.h``` for UART, I changed them by analogy with 6.1d, but after the firmware, the printer's behavior did not change... o_O...
* I probably missed something during this time and do not understand something....
* Would like to see a step-by-step instruction on adding the configuration of the VYPER-CE to a clean Marlin...;)

rommulaner, 2024-02-19:
It's not that easy and just copy the two configurations into a clean Marlin. The configuration files are changing from version to version and can't be copied. You need to create the configuration files step by step to fit the Marlin version. And then you need to implement the Display and all the other changes in Marlin that are needed. That's what I did for weeks...
The 6.1d in UART starts on my printer. But with the same settings in 6.2.1 it doesn't start.

KAPTOHKA, 2024-02-19:
* Yes, I know that need to edit and add a lot of files...
* In ``` Marlin-bugfix-2.1.x``` officially added support for ```Vyper & GD32```, I compiled (not UART) it and flashed it...,
the printer behaves exactly the same..., flashed the screen with native firmware and still the same behavior... and it's weird(((
* In short, there is no ready-made solution and need to go into the depths of the Marlin-code, but no one has done a roadmap and every time you have to study everything from the beginning (((
* Yes, 6.1d works, I previously modified it a little in [6.1e](https://disk.yandex.ru/d/ThRGO4p7Tg6lNg), but wanted to try ```I.S.```....

rommulaner, 2024-02-19:
The screen firmware must fit to the flashed printer firmware.

KAPTOHKA, 2024-02-19:
Yes...
I tried it on everyone, but it doesn't work...
[Even from here...](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/AnyCubic/Vyper)

KAPTOHKA, 2024-02-19:
v6.2.1-4 behaves the same as in the video above...

rommulaner, 2024-02-19:
Then I have no solution then using 6.1d that works on my printer with UART.

KAPTOHKA, 2024-02-25:
Maybe this is due to the fact that the bus UART is connected to the connector of the module WiFi, and the standard Marlin does not know about it and tries to configure it according to its standard scheme???
I read about it somewhere at **_Pmatsol58_**, but that was 2 years ago, now I can't find it...

rommulaner, 2024-03-09:
Added change to the main.
Thank You for the solution!

closed 2024-03-09 by rommulaner