rommulaner/Anycubic_Vyper_MB_CE_6.2 #25: Problems with the dimensional accuracy of printing in the Z direction

LeonardoHome, 2024-02-21:
I'm sorry I have to get in touch again. I have currently installed the latest version of the firmware, which fixed the problem with the component fan being too loud. 
Unfortunately, I now have another problem that I didn't have before.

I printed components that extend over more than half of the print bed and are exactly 6mm high. The first and all subsequent layers are 0.2mm high. Unfortunately, in reality the finished printed components are anything but exact in the Z direction. 
The 6mm high components are printed between 6.3mm and 6.8mm high. 

The Z axis steps have been calibrated and are perfect. So if I move the axis up 10mm and measure it, then in reality it has moved 10mm. The hardware and the control are therefore suitable. 

I also printed an XYZ calibration cube. There the deviation in Z is much more extreme. The cube should be 20mm high, but in reality it is 20.9mm high. 

I find this deviation from the target very worrying. 
I have also already tested two different slicers, cura and Orca. The result was that the slicers did not influence behavior.

The first layer, or the Z-offset, was set and is 0.2mm over the entire print bed (probably this is so good because of the 7x7 mesh). 

I'm at my wits' end and don't know what else I can check. 
Is it possible that there is a bug in the firmware, for example that the mesh is not being processed correctly? 

Thank you in advance for the help.

rommulaner, 2024-02-21:
What values do you have min max in the bed mesh?

LeonardoHome, 2024-02-21:
[image]

rommulaner, 2024-02-21:
There you have your deviation. 
Why didn't you run the z-probing with the HOME button on the leveling screen as written on the screen? Then the printer does a homing, heating an probes the Base Z-Offset. After that run the ABL and the values will be around 0.
The cause of the height difference is because of activated FADE HEIGHT of 10mm as standard setting to fade out the bedmesh from 0 to 10mm in Z height, to get flat parts at the top and your high bedmesh values.
The leveling procedure is also described in the display release notes.

LeonardoHome, 2024-02-22:
Thanks for the explanation. You're right, I've now done it as per your instructions and the result is now correct, thank you.

closed 2024-02-22 by LeonardoHome