rommulaner/Anycubic_Vyper_MB_CE_6.2 #28: Nozzle Temperature Drop after fans turn on

serioussam, 2024-03-01:
Hi, I have flashed the recent firmware(Feb 16th version ) and LCD to my Vyper. As I was doing the calibration I am having issues with hotend heating. Steps taken before the issue arose.

1. Tune Esteps.
2. Auto level bed.
3. Tune PID with following parameters , Temperature 205, Fans on, Cycle count 10.
4. Print point 8 cube for tuning the flow rate (Run into temperature issues with hotend).

My GCode for the print is set to 210 for first layer with no fans and Fans kick in for layer 2 onwards with temp as 205. I am using octoprint.

Layer 1 runs fine, Layer 2 as soon as fans turn on temps drop and continue dropping until the system restarts.

Attaching temp graph and error reported by octoprint. Let me know if you need more details. Next I will try to downgrade to see if older versions did not have this issue.
[image]
[image]
GCode: https://pastebin.com/tjmtpfsw

My Main motivation to install this was to get input shaping so I can improve on the ringing I have been seeing without having to go to klipper. Really hope this gets resolved.

rommulaner, 2024-03-01:
Do you have a silicone sock on the heatblock?
That would be my first improvement to avoid such behavior.
Your slicer settings are made fail on the system. Droping the temp at the same time when the cooling fan starts on full speed. That's the worst you can do. You can also start the fan gradually up to layer 3 or 4 or print with constant temp.

You can change the configuration and activate the auto fan reduction again, perhaps this can help. It reduces the fan speed at temp drops. But first you should have a proper hardware setup.

serioussam, 2024-03-01:
Though I agree with your suggestions as they are all good things to do. I have been printing the same profile on ages on both stock and later on Zombie Firmware. I did not have this issue till I flashed the Community firmware which leads me to believe it has something to do with how the firmware is handling this.  I will be putting on Sillicon Sock regardless (Arriving tomorrow). along with that running a test with constant temperature, And Temp change and different height to see if it changes anything.

serioussam, 2024-03-01:
I believe I figured it out why it worked for me before with stock or zombie firmware and why its not working now. Stock firmware has adaptive cooling turned on, while CE has it disabled. Let me test out with adaptive cooling on. Apologies if I was coming pretty hard. Been a frustrating experience trying to figure this problem out.

rommulaner, 2024-03-01:
Yes, how I already mentioned it. I deactivated it, because if the temp falls only slightly below the set temp, the fans are going down quite much. But part cooling is important when printing fast, that's why I deactivated it.

serioussam, 2024-03-01:
Makes sense, adaptive cooling is working now for me no more failed prints. Now I just need to test with sillicone sock to see if I can run this without needing to use adaptive cooling.

sdvileskis, 2024-03-09:
I had this issue trying to print a temp tower... First few layers were at a default 205C, but then the bottom section of the temp tower jumps up to 220C at layer 5 and the fan kicks on; Printer w/ CF6.2 would fail almost every time. I finally did a PID tune at 220 with the parts fan running,, reset the base temp to 210, and enabled adaptive cooling in firwmare and it finally was able to print a [terrible looking] cal tower. 
Ill add that mine was/is missing a silicone sock, previous owner said it never had one, and had the hotend replaced with a generic, so all bets are off regarding the thermistor (It seems to be reporting about 20C low 'at temperature')  .  Regardless, having the printer give up when it can't maintain temperature is extremely frustrating. I'd much rather have errors/warnings, not just aborting with a cryptic "I give up, please reset me like a Microsoft product" message.   Maybe do a pause/park instead so the user can decide to continue or abort the job?

rommulaner, 2024-03-09:
The firmware cannot see the reason of the failed temp regulation and if it the temp runs out of the configured borders it stops the print and gives the temp warning message. As already suggested in the other issue, I recommend using a silicone sock or gradually raise the fan speed. Also raising the watch time for the thermal runaway in the configuration_adv can give more room to avoid shut downs. I will also try to investigate the behavior on my vyper to see the effects.

sdvileskis, 2024-03-10:
It wasn't a thermal runaway code, it was a code for failing to maintain temperature, as in the temperature wavered too much. Running a PID autotune at a high temp with the fan running seemed to improve the situation, but a little sleuthing of some of the bad reviews on the Generic Vyper/Kobra Max hotends seems to suggest there's either 1) an issue with the thermistor contained in the generic hot ends (wrong or different), b) that the thermistor is not mounted properly for the size of the hotend hole (poor monitoring), or c) that the hotend isn't actually hot enough-strong enough to maintain temperature.  A common trend for the reviews was that they often didn't come with the silocone socks. 
Regardless, Being able to toggle adaptive cooling on the screen would be a welcome improvement.

7ApUn1c0r9, 2024-05-13:
the same error on versions 6.2.2 and 6.2.1-4. I also ordered a silicone sock for the hot end, but I noticed that when the PID is calibrating, the status bar says E0 heating, and when it prints E1 heating, could there be an error here?

rommulaner, 2024-05-13:
If you want, you can activate the auto fan slow down and compile it. With a silicone sock you won't have any problems. The message with E1 is not correct in Marlin as I saw. I will update that.

rommulaner, 2024-07-02:
Solved in release 6.2.2 with the versions with FSD (Fan Slow Down). There the fan will slow down if the hotend temps drops.

closed 2024-07-02 by rommulaner