rommulaner/Anycubic_Vyper_MB_CE_6.2 #29: undefined VIPER_NOZZLE_HOMING not working as expected, Leveling woes, Maybe some Temperature issues too

sdvileskis, 2024-03-09:
Greetings, and thank you for your work on this project! 
I've got 6.2 running on my 2nd-hand Vyper; I don't know the complete history of this unit, but a few things seem 'off' 
When my Vyper does a 4x4 ABL, there's roughly 1.5mm difference from the left column edges to the right edge. Maybe this is normal, but it seems a little extreme to me, like a z-axis sync issue. If I give one of my Z-axis screws a half turn or so, it seems MUCH better and skewing X back and forth is much more level. 

Even though VYPER_NOZZLE_HOMING is undefined, When I home via the 'HOME' button from the Auto/Manual Bed Leveling screen it still does a 0,0,0 home, but then it still seems to go to the center and reset the Z-offset (to the tune of about -2.5mm). Maybe that's by design, but my understanding from the comments in Configuration.h is that without VYPER_NOZZLE_HOMING, HOME would be done with the just the two z-stops, NOT the center bed test with heat.  

If I could just bring down one z-axis 1.5mm or so and it would otherwise be level. (My Vyper does not have adjustable Z-stops, and there isn't 1mm of adjustment in the optical sensor screws). Might it be useful to have independent Z control/offsets, or compensate for a consistently crooked bed by bumping one Z by a little bit? Does ABL use dual-Z control to compensate for beds skewed along the X axis/across the Y-gantry? smart when there's dual Z control?  I did see a Z2_ENDSTOP_ADJUSTMENT in Configuration_adv, but it of course was unit-less, so I don't know if it's in stepper steps, mm, a binary value, etc. I believe there is a marlin gcode for tweaking something related to this as well. Or If my Y-Gantry is really skewed that much, do I just need to disassemble and rebuild my machine with a square at the ready (everything seems pretty tight already)? 

Additionally, printing a bed level pattern seems like it disregards the -2.5mm home z-offset it determined from ABL screen homing, and on my printer, starts printing the pattern 2.5mm in the air! (is this a bug, or maybe an order of operations issue?) More concerning than that, attempts to cancel the bed level pattern from the screen didn't work, and it continued to print in air for almost a minute until I physically powered the machine off! (seems like buggy behavior!)

I saw a closed ticket for this, but with 6.2 I also noticed/suspect my temperature was a bit higher than the software believes. I tested it against an IR temperature gun, and it seemed the nozzle was roughly 10-15C higher than the printer was reporting. I had stringy messes for anything over "180C".  I do know that the previous owner swapped out the hotend with a generic vyper-compatible one, so have no idea if that temp sensor is correct or if there were variations across models. My question is this.. are there other thermistor settings that may be correct for the Vyper besides `5` ? Are there other board version settings (pull ups/downs) that might be impacting this? (The hotend and bed temp both seem OK and report ~19-21C at room temp)  I also know that my printer is missing the silicone sock, and that the parts fan coming on at layer 2 or 3 will cool the nozzle enough and lead to a failed print with a cryptic error that I suspect means "I can't maintain this temperature, so I'm giving up". I was able to get it a little better by doing a PID tune at higher temp with the parts cooler on, but it ultimately was still too hot. Should I adjust my hysteresis a bit so that it can take a bit more variation before aborting a print? (PLA has a pretty decent temperature range) 

Thanks again!

rommulaner, 2024-03-09:
Hey!
Sorry to say, but it would be much easier for me and others reading that, to open issues for each topic and not mixing them up in one. Thank you for doing that in the future.

As I understand your description your Z-endstops are not parallel to your bed, so you see a tilted bed in the mesh. At homing the firmware is compensating the tilt during homing, so that the endstops are triggered simultaneously.
I expect the hardware to be checked and ok, so no bend endstop finger that triggers the sensor a bit later or such things.
There you can adjust the tilt by writing M666 with the total tilt value. By using positive or negative values you can control the direction of the tilt. Just try and don't forget so save the settings and that this is lost after reset to firmware settings.

About your point of the homing. The homing button on the leveling screen is only to reset, home and measure the Base-Z-offset and not for normal homing. You should read the text above the button. 
If you just want to home you have to use the home button on the movement screen.

About the test pattern behavior you should do the Base-Z-offset homing, run the ABL and start the test print. There you can adjust the normal Z-offset if needed.
And yes, I also saw that the Cancel is not working there. I think that's a Marlin issue, but I have that on my list.

About the hotend temp I absolutely did't have any issues with that on my vyper. I print PLA with 215°C like my new K1 or Kobra 2 Max with no problems. There could be a difference in the sensor quality or manufacturer but the map is correct to this type of sensor. But in general that's part of Marlin itself and has nothing to do with CE firmware.

About the temp drop on the hotend I highly recommend using a silicone sock or gradually raise the fan speed in the slicer (like full speed at layer 4) as the ADAPTIVE_FAN_SLOWING is affecting the print quality too much in my opinion and also slows down the fans during the print with no changes at the temp. You can also adjust the Thermal Runaway settings in the configuration_adv to a longer time than the 60 seconds, that may help also.

sdvileskis, 2024-03-09:
Thank you for the reply- I'm happy to keep the scope of this issue to just HOMING related. I did a little more experimenting this morning and there's certainly an order-of-operations concern regarding homing, base-z homing, and ABL. 
a) If I do ABL without the base-z homing button (Z offset 0), ABL will get offsets on the order of -3.6mm (left side) to -1.0mm (right side).
b) If I do base-z-homing from the ABL screen first (Z~ -2.5mm), then do the ABL, it'll come up with numbers that are from -1.X to +1.# ;  
It's not entirely clear to me if a mesh with base-=0 (a) is equivalent to a mesh with  base-z=-2.# offset and resulting smaller mesh numbers (b). I did try a print test pattern and it printed a little bit, but ultimately a few sections ground the nozzle into the bed. 
Likewise, It's also not clear what will happen if:
c) One does a base-Z homing button from the ABL screen AFTER doing ABL 
I foolishly started a print following step c) and it promptly crashed my nozzle into the bed (and appeared to be about 2mm lower than the bed!). Additionally, trying to cancel the print seemed to just pause/park it! it took several cancel button presses over the course of a minute or two for it to actually stop, cooldown and return to the main screen. (is this a bug that deserves its own ticket, or an upstream Marlin issue?) 

If there is a required order of operations (motion screen homing, ABL base-z homing, and then ABL mesh generation), it seems there needs to be a feature to invalidate the mesh, disable it,or at a minimum, show a warning. (new ticket??) 
 
It's also not clear if ABL is actually being applied during a print from SD; Doesn't the stock firmware (or maybe base marlin) have a button to enable/disable the use of ABL? Is it safe to assume ABL is always enabled in CE62? (new ticket?) 

You mentioned testing M666; Documentation for it is not very clear, it just says "offset for Z axis endstops". What I think I need is an offset for just one of my two endstops. I did find and test the G34 command (Z steppers auto-alignment), which probes near the left Z axis, then moves to the right Z-axis. It initially detected Z2-Z2=1.4mm, and after a few iterations got it down to 0.06. It did seem to save something, and showed screen 61(saving) before SetNewScreen 28.  While it seemed like G34 identified and compensated for my ~1.4mm skew, running an ABL afterwards seemed to have the same values they've always had recently, with a consistent skew is Z from left to right.  Shall I try setting Z2_ENDSTOP_ADJUSTMENT to about -1.4, recompile, and see what happens?

rommulaner, 2024-03-09:
The suggested way to level bed properly is to push the Home-Button on the leveling screen to zero all offsets and probe the bed center to get the Base-Z-offset and start the ABL after that. That's also how it's written on the screen. Doing it the other way around will cause a crash, that's true, but making it fool proof will make the usability worse . It doesn't make sense to delete the mesh at the leveling homing, because if you change the nozzle or the steel plate for example you just need to do the leveling homing and can start to print without the need doing the bed leveling again.
You also don't need to do the movement homing at any point as it is already included in the leveling homing.
With the Base-Z-offset you just move the zero z-plane to the bed height in the center to get values around 0 over the whole mesh (if your bed is flat and parallel to X and Y). If you change the nozzle you can just do the leveling homing and bring the zero z-plane again to the tip of the nozzle and then also the mesh fits again.

Cancelling the a normal print work absolutely fine on my and other vypers. I have no idea why it shouldn't work on yours. 
Yes, the Cancel button of the mesh validation doesn't work for any reason, but I'm aware of that and I'm investigating that already.

The ABL is always activated during the print as standard unless you deactivate it in configuration or the start gcode.

The M666 is the right setting to adjust the endstop, that will work for you for sure. You just need to configure the correct value (positive or negative, depending on the tilt direction) and one z spindle will go up or down this value. If the value is correct, the gantry should the be parallel to the bed. So there is no need to change code and compile something.

sdvileskis, 2024-03-10:
Thanks-- I did an M666 Z1.7 (or something to that effect) and then did a base-z zero and repeated my auto-leveling. It seems to have made a significant difference. I'm not convinced my Vyper wasn't hit by a car in its previous life :-) 
[image]

I still maintain that if order of operations (base-z vs ABL) is important to successful leveling, it should be enforced in some capacity, even if only with a warning. I think if one is taking the time to chang a nozzle, it would warrant a full bed leveling sequence anyway, particularly since it only takes a minute or three and they're probably changing a nozzle due to contact with the bed. 

I haven't gone through a power cycle yet to verify everything I messed with is maintained, but I'm hopeful I can just power cycle and print. It would probably be helpful to have a dual-z alignment button for G34 leveling, or a M666 input, particularly for printers that don't have adjustable Z-stops. G34 seems to be pretty helpful, using the nozzle probe/strain gauge to actually measure the skew between the two Z-probes. I might try that again and see what results, though I admit I don't know the correlation between M666 and the results of the G34.

rommulaner, 2024-03-10:
The G34 is doing a dual z-alignment as you already mentioned, but it can't save that value anywhere, so it is overwritten at homing. It is normally used in the start gcode for printers with only one endstop, but as the result is connected to each probing, the result will vary a bit at each run. That's why you should fix the bed mechanically first and as last option to set the value permanently with M666. For me it looks like something has fallen on the bed and bent it. I also think that the bed leveling can't fix that bent bed good enough as it is only probing 16 points and you mentioned that the nozzle is touching the bed on some points.

Adding such a Endstop adjustment is possible, but at the moment I don't think it is really necessary and will cause more problems than improving the usability.

sdvileskis, 2024-03-12:
Somewhat related to this ticket.. if i try updating babystep-Z from the touchscreen while it's printing, it doesn't seem to change the printer. The touchscreen will reflect the change, but even an extreme change like 1mm doesn't seem to move the nozzle in Z. 
If I cancel and start another print, then it seems the z-offset is applied.
It didn't seem like it was dragging the nozzle bad, but it seemed to (try and) print at least 5 layers atop one another without any observable Z-change (the LCD status was incrementing reported increasing Z just fine), perhaps because my combined offsets and bed leveling are off by -1 to -3mm combined, it was disregarding the babystep Z I entered or adding it at the wrong place.  I'd expect runtime tuning of babystep to be added to current position regardless. After the futile effort to extrude ~2-3mm into the first mm or so, it finally started actually climbing in Z and the print looks fine.

rommulaner, 2024-03-12:
I would bet that it was connected to USB while you did this.
Try without it. I think the USB connection leads to conflicts in the command chain of the different sources and the display commands are not the first to be considered.
I don't know why, but it looks like that. 
You need to think like that: This firmware is not completely new and it is running fine on many other printers because the main code ist several years old and tested.

rommulaner, 2024-04-02:
Babystep works absolutely fine on mine and every other vyper. Perhaps you messed up the calibration procedure.

1. Go to calibration, bed leveling and click on the home button to heat up and probe to the bed center and see a value for the base z-offset.

2. Then just start Auto Bed Leveling with the button below. It applies the base z-offset while homing and z=0 is now bed height at center. During leveling you should see values around 0.000. If there are high values check rollers, look for loose screws or flatten the bed by bending it. Let ABL finish.

3. Now you can start a test print and adjust the babystep z-offset by clicking on the z-offset at the print screen or on the tune button and adjust it with + or - by 0.01. At every click it's doing 0.01 adjustment of Z and displays the change only if it was applied. You can normally also feel or hear the application of the babysteps on the printer. Normally you need to babystep up to +0.10 or a bit less to get a perfect first layer on the textured PEI plate. That's it.

The saved base z-offset and babystep z-offset are always applied when it's doing the G28 homing on z with a stock hardware setup (2 Z endstop switches).

So just click, click and babystep. Don't change anything by hand or do something else. It's so easy and works for shure.

rommulaner, 2024-07-02:
As there are no more updates, I close this issue.

closed 2024-07-02 by rommulaner