rommulaner/Anycubic_Vyper_MB_CE_6.2 #30: Cancel Print from SD card w/ 6.2 doesn't always cancel

sdvileskis, 2024-03-10:


sdvileskis, 2024-03-10:
Additional Info, Reviewing my Configuration.h changes from the git head... Nothing I've changed jumps out... 
I overrode the Machine Name (so I knew that the updated config was applied); Set PID fields Kp,Ki,Kd, Reduced MINTEMP to from 170 to 150, Added 3rd PREHEAT profile, Defined NOZZLE_CLEAN_FEATURE, PRINTCOUNTER, Adaptive Fan slowing,  PID FAN SCALING, and backed HOTEND_IDLE_TIMOUT_SEC from 15min down to 5 min

rommulaner, 2024-03-10:
There I have absolutely no idea why the Stop button should not work. It is a very simple operation which just sends a Cancel_Print command. It also worked every time on my vyper and was never changed since the CE exists.
Try to reflash the screen and mainboard and reset the settings several times. Sometimes the display behaves strange after flashing or messed up some data.

KAPTOHKA, 2024-03-10:
> Try to reflash the screen...

By the way, in your screen firmware, files 13 and 14 are named in Chinese, some screens do not flash such files and after that, the screen works with glitches ...
You need to rename these files to `13TouchFile.bin` & `14ShowFile.bin` and flash them again.
You can check after the firmware, on the blue screen, that the files have been flashed...

rommulaner, 2024-03-10:
In the release ZIP the files are correctly named. In the SRC folder the file names are different, that's why there is a build-script.

sdvileskis, 2024-03-11:
> In the release ZIP the files are correctly named. In the SRC folder the file names are different, that's why there is a build-script.

OK, thanks I'll check it out.. In whatever zip I had w/ LCD 6.2 when extracted, the path names became part of the filenames; it was really awkward; I had to rename them all because they had DWIN_SET as part of the filename.

sdvileskis, 2024-03-11:
After a touchscreen flash, I thought it was fixed, but I was mistaken.

Just replicated it twice..
Stop -> Stop Current Print? Confirm -> Returns to Home Screen w/ "Print Aborted" -> Nozzle moves home. 
Almost immediately after reaching home, screen shows "Parking" then "Nozzle Parked", and goes back to Printing Screen -> Print Paused -> Heating -> 150C.  (timer shows 7:20s) 

(2nd press) Stop -> Stop Current Print? Confirm-> Returns to Home Screen w/ "Print Aborted" again. (Nozzle already home), Immediately goes back to Parked screen and continues "Heating" 

(3rd press) Stop -> Stop Current Print? Confirm -> Target temp drops to 0C and it returns to home screen. 
https://photos.app.goo.gl/EHgsqCZTdwKymsCd9

rommulaner, 2024-03-11:
I also saw in your video that is connected to something over USB. Can you try stopping the print without the USB connection?

sdvileskis, 2024-03-11:
UPDATE: In 100% of my 1 recent tests, Unplugging the USB seems to have resulted in a proper abort on press of the LCD cancel button. (Octoprint had been polling status)

closed 2024-04-01 by rommulaner