rommulaner/Anycubic_Vyper_MB_CE_6.2 #34: Upgrading from CE-V6.1e

MarcKarasek, 2024-04-21:
I have a couple of Vypers running the CE-6.1e(LA) FW.

I see in your release you have a LA 4x4 and LA 7x7 what are the differences between them?  Are either equivalent to the 6.1e(LA) version?
Would like to update both printers or at least one of them.  But don't fix what is not broken comes to mind.. 

Marc

rommulaner, 2024-04-21:
Hi!

It described in the release notes. 
4x4 and 7x7 is the bed mesh size.
You can always downgrade if you need to. The 6.2 has Input Shaping.

MarcKarasek, 2024-04-21:
So 6.1e has a 4x4 as I see 16 total points for bed leveling..  Am I correct?

rommulaner, 2024-04-22:
The 6.1e was not done by me.
Yes 4x4 has 16 probe points. In my opinion you don't need more than that.

MarcKarasek, 2024-04-24:
Just upgraded one of the two.. went ok.  
One question.. your FW image is about174K in size the previous one I flashed 6.1e was quite a bit bigger at 478K. 
Do you know why?
Just wondering.

closed 2024-04-24 by MarcKarasek

rommulaner, 2024-04-24:
That's a good question. I have no influence on the file size. That's what the Marlin Builder is compiling out of the used Marlin Version.

MarcKarasek, 2024-04-24:
It works so ... just a ciriosity. I guess