rommulaner/Anycubic_Vyper_MB_CE_6.2 #35: Error: Build environment 'trigorilla_v006' is incompatible with BOARD_AC_TRI_F1_V1. Use one of these: STM32F103RET6_ac_tri_f1, STM32F103RET6_ac_tri_f1_maple

mat-pandaz, 2024-04-21:
Hi I'm really sorry - I've never used platform.io or know much about this, I'm just trying to update my vyper.  I installed the VS Code platformio extension and clicked build but I get this

 Executing task: C:\Users\matpa\.platformio\penv\Scripts\platformio.exe run 

Processing trigorilla_v006 (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Build environment 'trigorilla_v006' is incompatible with BOARD_AC_TRI_F1_V1. Use one of these: STM32F103RET6_ac_tri_f1, STM32F103RET6_ac_tri_f1_maple
=============================================================================================================================== [FAILED] Took 1.19 seconds ===============================================================================================================================

Environment      Status    Duration
---------------  --------  ------------
trigorilla_v006  FAILED    00:00:01.189
========================================================================================================================= 1 failed, 0 succeeded in 00:00:01.189 ========================================================================================================================= 

 *  The terminal process "C:\Users\matpa\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

rommulaner, 2024-04-21:
Hi!
Do you want to make changes in the code or why do you want to compile it?

mat-pandaz, 2024-04-21:
Hey - no I just want to flash it to my Vyper, It has all stock components so I made the change to the version as per your readme.  But it says you can't distribute the zip so I'm just trying to build it as it comes so i can flash my vyper.

closed 2024-04-21 by mat-pandaz