rommulaner/Anycubic_Vyper_MB_CE_6.2 #36: Z Homing

Gawper5, 2024-04-23:
Whenever i home Z, it gets wrong number example zOffset = -4.5, but when you start the print the nozzle start too low so i have to have zOffset = -2.5

rommulaner, 2024-04-23:
What version do have on the printer and screen?

Gawper5, 2024-04-23:
latest screen
printer LA IS 7x7 300°C

rommulaner, 2024-04-23:
6.2.2 is the newest version, currently on Prerelease but is nearly similar to 6.2.1-4 what is the latest release.
Ok. How was the leveling done? 
Did you probe the base z-offset with the home button on the calibration screen before running the ABL?

rommulaner, 2024-04-23:
That's the problem: 
1. Base Z-Offset probing 
2. start ABL

Otherwise you double the offset.
It's also written in the release notes of the screen and also on top of the calibration screen.

Gawper5, 2024-04-23:
Looks like it work now. Thank you

closed 2024-04-23 by Gawper5