rommulaner/Anycubic_Vyper_MB_CE_6.2 #39: No error message for hotend temperature abnormal

frzme, 2024-05-08:
My printer has issues with the hotend thermistor. With the original firmware it eventually shows the error message "ntc hotend temperature abnormal" with the community firmware it just reboots without showing any error message or clear indication what may be the problem

rommulaner, 2024-05-09:
Hi!
What version are you using? 4x4 with standard Hotend?

frzme, 2024-05-09:
6.2.2
7x7 with Standard Hotend

rommulaner, 2024-05-09:
Can you test the attached MB firmware and give me feedback if it now shows the message correct.
[main_board_20240509-155833_7x7_TEST_MSG.zip](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/files/15263244/main_board_20240509-155833_7x7_TEST_MSG.zip)

frzme, 2024-05-09:
Tested but unfortunately did not succeed. Out of 3 attempts it got stuck on the loading screen after a reboot 2 times and once it completed the reboot.
[image]

rommulaner, 2024-07-02:
Any news here?

frzme, 2024-08-24:
My last post meant to say that I tested the firmware build you provided but the hotend abnormal caused a reboot and not an error message.

I've since fixed my hotend so I hope I won't be needing the message again soon