rommulaner/Anycubic_Vyper_MB_CE_6.2 #40: Media Init Fail

Darkboyzero, 2024-05-18:
After replacing the memory card, this message is stuck. It does not disappear even during printing. I have tried to replace it multiple times, but the following message remains.
" Media Init Fail" 
[image]
[image]

rommulaner, 2024-05-18:
That's an error message and doesn't disappear by itself. After restart it's gone.

Darkboyzero, 2024-05-18:
After restart it's gone. I change the memory card again and it stays there.

rommulaner, 2024-05-18:
If you pull the card during reading it drops that message. Normally this occurs when you are in the file menu.

Darkboyzero, 2024-05-18:
6.2.14/ 6.2.2 version message is stuck .

rommulaner, 2024-05-18:
What do you mean?

Darkboyzero, 2024-05-18:
I'm in the home menu, I take out the memory card, upload the stl file to it, and then put it back. And the message sticks.

Darkboyzero, 2024-05-18:
I tried them, this is the error for all versions.
[Release Vyper-CE-6.2.2 MB Firmware](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/releases/tag/Vyper-CE-6.2.2)
[Release Vyper-CE-6.2.1-4 MB Firmware](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/releases/tag/Vyper-CE-6.2.1-4)

rommulaner, 2024-05-18:
STL files on the card?!
That won't work for sure...

Darkboyzero, 2024-05-18:
It works, printing starts. The message still sticks.

Darkboyzero, 2024-05-18:
Insert the memory card. Take it out . The message appears. Reload it, the print menu will appear. Start printing. Printing goes on, message remains. " Media Init Fail " .

rommulaner, 2024-05-18:
I think you print a gcode file.
And yes, the message will stay as long as the is nothing else to display. With no preheating it should show bed heating and hotend heating actions. If there is M117 in the gcode, it should display that.

Darkboyzero, 2024-05-18:
Normally, the following message appears. E- Heating .... & xxxx .gcode . It doesn't change, the message gets stuck . " Media Init Fail" .

rommulaner, 2024-05-18:
If there is initiation error the message will stay, that's how it should be. Maybe to many files on it, too long file names, symbols in the filename, some bad sectors on the card, wrong format or incompatible card.

Darkboyzero, 2024-05-18:
This version works fine. [Release Vyper-CE-6.1d MB Firmware](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/releases/tag/Vyper-CE-6.1d) . The message does not get stuck when I change the memory card.

Darkboyzero, 2024-05-18:
Sorry , "Media Init Fail"  & print & next message " E -heating" & next message 10x10x2_PLA_18_1s.gcode . These versions do not switch. The message remains, it does not move on to the next message. Buggs messege " Media Init Fail" . Remove and reinsert the memory card several times in succession. Then the message will stick.

rommulaner, 2024-06-14:
Maybe, depends on the benefits. Currently there is nothing that's need to be fixed.

Darkboyzero, 2024-06-17:
I deleted the message, it works perfectly fine.
original : 
  else {
    #if EITHER(HAS_SD_DETECT, USB_FLASH_DRIVE_SUPPORT)
      if (marlin_state != MF_INITIALIZING) LCD_ALERTMESSAGE(MSG_MEDIA_INIT_FAIL);
    #endif

new: 
  else {
    #if EITHER(HAS_SD_DETECT, USB_FLASH_DRIVE_SUPPORT)
      if (marlin_state != MF_INITIALIZING);
    #endif
I don't get a message, but I don't miss it either. :D 
Sorry if I don't understand, but I'm trying to communicate with google translator. I only understand basic words that are needed for configuration.

Now that it's finally working, I'll start testing the "input shaping" option.
I reset the acceleration to the factory value, but I don't print fast anyway.
I wonder if it can eliminate ghosting.
Thanks for your work, it meant a lot to me on my machine.

rommulaner, 2024-06-18:
That's one way to do it.
But can you try LCD_MESSAGE instead of LCD_ALERTMESSAGE and give me feedback?
Then it should be only a normal message that can be overwritten by other messages, that would be a better solution.

So it should be:

if (marlin_state != MF_INITIALIZING) LCD_MESSAGE(MSG_MEDIA_INIT_FAIL);

Darkboyzero, 2024-06-19:
I try. Thank you very much .

rommulaner, 2024-07-02:
Changed to normal message with release 6.2.2.

Darkboyzero, 2024-07-02:
Thanks, I tested it and it works.

Darkboyzero, 2024-07-03:
Hello , Factory Default Settings does not work. The button beeps, but does not return to its original position.

rommulaner, 2024-07-03:
It's set to hold for 2 sec. to avoid accidentally reset.

Darkboyzero, 2024-07-03:
Thank you, yes it works.