rommulaner/Anycubic_Vyper_MB_CE_6.2 #41: Question about input shaping

GremlineQ, 2024-05-20:
I have no idea where to ask, so I'll ask here if it is allowed. As far as I know, this firmware is based on Marlin, and I just want to know how to connect accelerometers so that I can calibrate the input shaping, because from what I know, it could be connected to Raspberry Pi in Klipper, but how should they be connected in Marlin? And is it worth calibrating input shaping in this printer?

convicte, 2024-05-20:
https://marlinfw.org/docs/gcode/M593.html

More specifically:
[image]

rommulaner, 2024-05-20:
Exactly like it's been written in the post above. You need to do a test print like you can read in the link.

closed 2024-05-20 by GremlineQ