rommulaner/Anycubic_Vyper_MB_CE_6.2 #42: firmware with enabled skew correction

GremlineQ, 2024-06-30:
I think that a good idea (if possible) would be to enable the skew correction function in the already prepared firmware files, especially now that califlower is free to download for a week, many people may be interested in it.

rommulaner, 2024-07-02:
What should there be activated? I think it's active by default.

GremlineQ, 2024-07-02:
> What should there be activated? I think it's active by default. 

from what I know so far, the command for skew correction is m852 in the console, when I tried to use it, the output of this command was an unknown command

rommulaner, 2024-07-02:
Ah, I see, it's deactivated...
What is there a sufficient way to configure it? [image]
Any suggestions?

GremlineQ, 2024-07-02:
ok, sorry for the trouble, I thought that it would be enough to turn it on, but from what I see, the values that will be calculated from the test must already be entered there (i.e. they may be different for each printer), so probably people who would like to have this skew correction would have to build the firmware themselves

rommulaner, 2024-07-02:
What I just read about the califlower it works with the factors. So it's easy to enable and set the factors to zero and also enable the M-command. 
I can compile a version for you to test it. What variant do you need? 4x4 with stock Hotend and FSD?

GremlineQ, 2024-07-02:
1) I use 4x4 with fullmetal hotend 
2) thang (just now I saw that this promotion has ended (if you want I can send you the filles))

convicte, 2024-07-02:
Two words - intellectual property!

rommulaner, 2024-07-02:
Here the 4x4 AllMetalHotend with activated skew.
Please test and give me feedback if it works as it should.
[main_board_LA_IS_4x4_AllMetalHotend_SKEW.zip](https://github.com/user-attachments/files/16071911/main_board_LA_IS_4x4_AllMetalHotend_SKEW.zip)

GremlineQ, 2024-07-04:
works without any problems

closed 2024-12-01 by GremlineQ