rommulaner/Anycubic_Vyper_MB_CE_6.2 #43: Vyper crashes after 2-3 layers

EdenJeong1, 2024-08-03:
Things I have did:
Switched slicers from Cura to Orca to no avail
Flashed firmwares back and forth.

It may be a g-code error but don't know what to do.

rommulaner, 2024-08-03:
Maybe a faulty SD card or a command in the gcode that is not supported by the printer.

EdenJeong1, 2024-08-03:
I've tried just about everything including the ones you said.. what other
options can I do next? I tried slicing in different software but did not
work

EdenJeong1, 2024-08-04:
Error:Thermal Runaway, system stopped! Heater_ID: E0
Error:Thermal Runaway, system stopped! Heater_ID: E0

app fs start: 0
mount: 0:/
start
 Watchdog Reset
Marlin Vyper-CE-6.2.2(LA+IS)
echo: Last Updated: 2024-01-21 | Author: (Rommulaner, Vyper CE-6.2.2 Marlin
2.1.2.1)
echo: Compiled: Jun 29 2024
echo: Free Memory: 43543  PlannerBufferBytes: 3584
Loading DWIN LCD setting from EEPROM
echo:V88 stored settings retrieved (917 bytes; crc 6758)
//action:notification Stored settings retrieved
Taring probe
//action:prompt_end
echo:SD card ok
Error:Line Number is not Last Line Number+1, Last Line: 0
Error:Line Number is not Last Line Number+1, Last Line: 0

Pronterface logs. Can't be a SD card error if its connected directly from
usb

rommulaner, 2024-08-04:
If it's a problem with a temperature drop while the fans kick in, you can switch to a variant with FSD. This slows down the fans when the temp is dropping.

closed 2024-08-25 by rommulaner