rommulaner/Anycubic_Vyper_MB_CE_6.2 #45: Bed Levelling Does Not Complete

b00sted03, 2024-10-09:
Any way to configure all-metal/300deg C limit on 6.1.8d?  Having major issues using ABL on 6.2

rommulaner, 2024-10-10:
You should read the Release Notes at the screen release. You need to probe the Base Z-Offset first (home button in the calibration menu) and start the ABL after that. Then everything should run perfectly and the values are then around 0.
There is also no need to compile the fw as it is ready to flash in the Release section to download.

b00sted03, 2024-10-10:
Printer comes up, I home, it homes fine (this is done in the calibration screen), I ABL, it starts running G34 iterations

As above, this is what I am doing.  I have read your release notes and am following the instructions.

b00sted03, 2024-10-10:
I will try to download the pre-compiled version for the screen, try again and provide a video link if its behaving similarly

rommulaner, 2024-10-10:
When you tap the home button on the calibration screen the printer should be homing and probe the center of the bed after heating up and give back the probed value. Then you can start the ABL.
Would be good to see a video of the whole process to see what's happening.

b00sted03, 2024-10-10:
Yeah can confirm that's what's happening (prior to this convo).  I'll get this flashed and update you.  Sidebar, do you have a donation link?  I do very much appreciate all the work you've put into this and can see the improvements already.

rommulaner, 2024-10-10:
Yeah, you donate with PayPal to @robertrwwagner if you want.
Thanks 😃

closed 2025-01-20 by rommulaner