rommulaner/Anycubic_Vyper_MB_CE_6.2 #47: BLTouch supported?

dr-pavel, 2024-11-01:
Trying to build firmware with BLTouch option enabled, compilation stops with an error:
```
Compiling .pio\build\STM32F103RET6_ac_tri_f1\src\src\HAL\STM32\timers.cpp.o
Marlin\src\HAL\STM32\timers.cpp:328:40: error: static assertion failed: One or more timer conflict detected. Examine "timers_in_use" to help identify conflict.
  328 | static_assert(verify_no_timer_conflicts(), "One or more timer conflict detected. Examine \"timers_in_use\" to help identify conflict.");
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~^~
*** [.pio\build\STM32F103RET6_ac_tri_f1\src\src\HAL\STM32\timers.cpp.o] Error 1
```
Here is part of my configuration:
```
// @section info

// select build type here
//#define VYPER_BUILD         // standard
//#define VYPER_BUILD_IS      // standard but with input shaping
//#define VYPER_BUILD_LA      // with linear advance and junction deviation enabled
//#define VYPER_BUILD_LA_T    // as above but with uart connection to TMC2209's for x, y, z and z2
#define VYPER_BUILD_LA_TE   // as above but with software serial connection to e stepper
//#define VYPER_BUILD_LA_IS   // with linear advance and junction deviation enabled and input shaping
//#define VYPER_BUILD_LA_T_IS    // as above but with uart connection to TMC2209's for x, y, z and z2
//#define VYPER_BUILD_LA_TE_IS   // as above but with software serial connection to e stepper

#define GRID_MAX_POINTS_X 7   // moved here for easier change of the bed leveling mesh size

#define ALL_METAL_HOTEND      // activate if all metal hotend installed; hotend max temp will be raisded from 260 to 300

#define SLOWDOWN_FAN_ON_TEMPDROP    // activates ADAPTIVE_FAN_SLOWING in configuration_adv.h

```

It doesn't matter which configuration is active, if BLTouch is enabled, a compilation error appears.
I was planning to install a BIQU H2 V2S extruder with BLTouch support, but I had a problem.

rommulaner, 2024-11-01:
You need to change the lines 131 to 133 in the buildroot\share\PlatformIO\variants\MARLIN_F103Rx\variant.h to

#ifdef TIMER_SERVO
#undef TIMER_SERVO
#define TIMER_SERVO TIM2 // TIMER_SERVO must be defined in this file
#endif

I don't know if the TIMER 2 is the right one, but it should be correct as it compiles.

dr-pavel, 2024-11-01:
A huge thank you for your help. My knowledge is not enough to solve such problems (I am not a programmer), the maximum I can do is to configure the firmware.
Now the next printer upgrade is coming :-)