rommulaner/Anycubic_Vyper_MB_CE_6.2 #48: Bed Leveling / Z-Offset Stuck

FunkyOwl-Net, 2024-12-15:
Today i did some hardware renewels and wanted to update to your newest version. The flash was succesfull and like described, i did the PID-Tuning first. After that i wanted to do the automatic bed leveling and therefore i wanted to adjust the Z-Offset like i should. The temperatures rises and all looks fine but the display is stuck at "Homing and Heating" and no matter how long i wait, i cant adjust the Z-Offset.

I did also a reflash, no changes. Also i found out, that i cant to a factory reset. I can use the button, but nothing happens.

What can i do?

rommulaner, 2024-12-15:
The PID Tuning of bed and nozzle are important for the firmware to hold the set temps properly. 
You don't have to adjust the Base Z-Offset, it's done automatically when you press HOME in calibration menu. Then it's homing, heating up nozzle and bed and after stabilization it probes the bed. After that you get the probed value in the menu which you can adjust, but why?!
After that you can start the Auto Bed Leveling.
If it doesn't work like that, please make a video of the whole process for a better understanding.
The RESET Button is set to HOLD 2 sec. for action to avoid unwanted resets.

FunkyOwl-Net, 2024-12-15:
I did the PID-Tuning at the first start. Also now i did the factory reset and the PID-Tuning again, entered the ABL-Menu and pressed the HOME-Button, after that the Bed and also the Nozzle were heated up and i'm stuck at this stage, please see the picture.

Before the Update everything was fine with the last version!

Nothing happens, only a restart helps to get out of this screen.

[image]

rommulaner, 2024-12-15:
So, you have flashed Mainboard and LCD with the latest versions?
What version did you use before?

FunkyOwl-Net, 2024-12-16:
Yes, i flashed the newest version and also the newest LCD Version. I can't remember, which version i was running before but if i remember right, it was the 6.2.1-2 or Vyper-CE-6.2.1-4. As i said, before the update, everything worked awesome. 

I tried to flash also the 7x7 version, no changes. Also if i reboot the printer, the automatic Z-Offset isnt saved. If i start the ABL it will only check one point and then all the points have 0.000 and will be saved also like that.

rommulaner, 2024-12-16:
Before you wrote that the automatic Z-Offset doesn't work and now you wrote it doesn't save the value?!
You need to describe detailed what you do and what happens in order to get help.
I would suggest that you flash back to the version you had before and check if it works. Maybe you messed up something during your hardware changes.

FunkyOwl-Net, 2024-12-16:
Yes, both problems are here. I tried to do the ABL although the automatic Z-Offset didnt work because i wanted to try, if the automatic Z-Offset will be saved anyway and i'm only stuck at the screen, thats why i tested.

The Hardware changes shouldnt be the problem because i only changed the belts, the rolls, hotend and cleaned up everything so the printer is still stock but all the parts are renewed and cleaned.

Thats what i did so far: 
Flashed the new Firmware and LCD Firmware (also tried other SD-Card) -> After sucessfull flash i did a factory reset -> then i did the PID-Tuning -> then i wanted to do the automatic Z-Offset which didnt work -> for testing purposals i tried to do the ABL which also didnt work

So far thats all i did.

I will try to flash an old version to have a look if its working again.

rommulaner, 2024-12-16:
For me it looks like something is triggering the load cell in the print head while it's moving in Z or the load cell or wiring got damaged at your maintenance. Everything else works fine except the probing.
And this firmware works also fine on many other vypers.

FunkyOwl-Net, 2024-12-18:
Sorry for my late response, the last days were a bit busy. What do you mean by load cell and how can i find out if this is the issue. Because before i want to downgrade the firmware, i want to check if this is the issue. I checked all the wiring and also the pressure sensor is only lighting, if the nozzle touches the bed.

rommulaner, 2024-12-18:
The load cell is the sensor for probing installed in the print head which lights up when you touch the nozzle. You need to make sure that nothing is touching the Bowden tube when it's probing. Also the cable harness to the print head can trigger the sensor. Also the print head cover is very close to the hotend and can trigger the sensor. Maybe the cable of the sensor is broken in the wiring. I don't think it has something to do with the firmware.

FunkyOwl-Net, 2024-12-19:
I checked everything now. The loading cell is working fine, it is very sensitive but works fine, its also quite new (less then 6 months). The red light from the sensor is also only lighting, when touching anything. To be save ive removed the print head cover with fans in it and also the bowden tube. Also the cables from the hot end are not disturbing the load cell.

I downgraded to different CFW's but no changes, its not possible for me to do the probing or the automatic Z leveling. Then i reflashed the latest official firmware from anycubic homepage, there automatic probing was fine and also printing works.

So for now i dont know what to do next. I checked all wirings, changed to another new hotend, removed every part which could interact with the load cell, nothing changes. I'm Stuck in the state i uploadet the screenshot before.

I also buyed a new SD-Card with only 4 GB's, and saved the files from 2 different PC's and different card readers only for testing, also no changes. I formated everything like described.

rommulaner, 2024-12-19:
Ok, so the probe itself seems not to be the problem.
Can you please describe what the printer is doing when you tap the home button on the calibration screen?
It has to do a homing and move to the middle of the print bed and start to heat up nozzle and bed. After stabilization of both temps it should start probing once and place the nozzle to Z0 so that the User can check the position and adjust it if needed.

rommulaner, 2024-12-19:
As i can see in the video everything is working as it should until the probe gets triggered by the nozzle touching the bed. At that point the nozzle should back off some mm and then return to the probed height. It might can have something to do with allowed range of Z-height for probing. Can you connect the printer to your PC with pronterface and repeat this probing to see what value it gives back for the probing. Then it should be a simple fix.

FunkyOwl-Net, 2024-12-19:
Thanks for Info. I tried with prontaface:

`Connecting...
Printer is now online.
Invoking handler for screen 43with VP=4166 value=4
SetNewScreen: 61
SetNewScreen (saving): 61
SetNewScreen: 43
SetNewScreen: 42
SetNewScreen: 28
Invoking handler for screen 28with VP=4098 value=2
SetNewScreen: 42
SetNewScreen (saving): 42
SetNewScreen: 28
Invoking handler for screen 28with VP=4098 value=4
SetNewScreen: 80
SetNewScreen (saving): 80
SetNewScreen: 52
SetNewScreen (saving): 52
Invoking handler for screen 52with VP=4164 value=1
echo:enqueueing "G28 U0"
echo:enqueueing "G0 Z5 F240"
echo:enqueueing "G0 X125 Y127 F4800"
echo:enqueueing "G30"
SetNewScreen: 61
SetNewScreen (saving): 61
SetNewScreen: 52
SetNewScreen: 61
SetNewScreen (saving): 61
Wait for bed heating...
Taring probe
Taring probe
Error:Probing Failed
Error:Probing Failed
`

rommulaner, 2024-12-19:
Hmm, PROBING FAILED is not that much information...
How much about is the distance between nozzle and bed when it's waiting for bed heating?
It should be around 5mm.

FunkyOwl-Net, 2024-12-19:
Its about 2-3mm.

rommulaner, 2024-12-19:
You can try to adjust the offset manually and try if then the bed leveling starts. You need to home the axis first, then go to the calibration menu and with +/- you can raise the value to about +2.5mm. Then try to level the bed. If it runs through that it's a problem with your too low nozzle. 
Is it true that after homing and moving z to 0 it will crash into the bed?

FunkyOwl-Net, 2024-12-19:
Yeah, now the bed leveling starts, thats nice. Yes its true that the nozzle would crash into the bed after homing. So the issue could be solved if i adjust the endstops, right?

Im printing these now on my kobra maybe it will help:

[image]

Othern than that i have a question: May i ask also, ive read that you are printing with 150 mm/s, right? Is your Vyper stock or do i need to do some hardware upgrades to gain also that printspeed?

rommulaner, 2024-12-19:
Great! 
Yes, adjusting the end stops will sort out this issue, but adjust them correctly because the z spindles are aligned with the two Endstops and then maybe your x axis isn't parallel to the bed and in the bed leveling it appears as a tilted bed from left to right.
Yes I print with that speeds but only inner walls an maybe infill. Outer walls are at 100 I think. My Vyper is completely stock except of the silicone sock an the firmware :) 
In the video I also saw that your nozzle is sticking out quite far from the heatblock. I think that the heatbreak has a longer thread than it should have here. May that's the root cause here. For a better heat transfer it should be nearly be completely in the heatblock.

FunkyOwl-Net, 2024-12-19:
Thanks for helping me out, now it works like a charm and i can adjust the bed with my dial gauge. It would be also great if this Firmware would work with my Anycubic Kobra, do you think i can also parse it for that printer?

Great Work, Great Help. Do you have any Paypal Link so i can give you a little tip?

rommulaner, 2024-12-19:
Great to hear!
As far as I know the Kobra is also based on Marlin and the Display is the same I think. Then it should be possible in some way. Also the Kobra2 with Marlin has a setup like that. With some work it should be possible.

rommulaner, 2024-12-19:
If you want to donate something I would be very happy :) 
PayPal @robertrwwagner 
Thank you!

rommulaner, 2024-12-26:
Closed as there is no issue any more.

closed 2024-12-26 by rommulaner