rommulaner/Anycubic_Vyper_MB_CE_6.2 #49: Jerk values can't be set per-axis

kb-elmo, 2024-12-23:
Changing the Jerk value of one axis through the movement settings on the screen automatically changes the value for all axis which should not be the case since the different axis have different ideal Jerk values.

For example when trying to set the Jerk value for the X-Axis to 8, the value for the Z-axis ends up getting changed to 8 as well.
It should instead be possible to set them to different values since Marlin supports that.

[image] [image]

rommulaner, 2024-12-23:
As all variants of the 6.2.2 are configured with active Linear Advance there is only one Jerk value used for all axis. The Classic Jerk configuration doesn't give good print results, that's why we use Linear Advance.

kb-elmo, 2024-12-23:
> As all variants of the 6.2.2 are configured with active Linear Advance there is only one Jerk value used for all axis. The Classic Jerk configuration doesn't give good print results, that's why we use Linear Advance.

Linear Advance is not a replacement for Classic Jerk. 

That's what Junction Deviation was meant for.  
But it's not really compatible with Linear Advance and somewhat problematic anyway.

Linear Advance is merely working on top of the Classic Jerk system by improving the extruder acceleration.  
It does not directly affect the cornering velocity of the other axis which is the main point of the Jerk settings.

And Linear Advance still respects the individual Jerk values for the extruder as well as per the Marlin Docs

https://marlinfw.org/docs/features/lin_advance.html
> LIN_ADVANCE now respects hardware limitations set in Configuration.h, namely extruder jerk

Marlin still requires individual Jerk values to be set for all axis in Configuration.h unless Junction Deviation is enabled (which it shouldn't when Linear Advance is used).
Setting them all to the same value makes no sense and there is no point in the Marlin docs where it says that you should do that.

rommulaner, 2024-12-23:
Would be interesting to see if classic jerk will bring some improvement together with LA. You can compile yourself a version and test it. If it works well, we can add it as a variant.

kb-elmo, 2024-12-23:
Will do :)

kb-elmo, 2024-12-24:
Left one was with Junction Deviation (firmware downloaded from the releases), right one is with Classic Jerk.  
Both used the same Linear Advance value and are printed in PLA (same manufacturer, just different color).

The right one has significantly less overrounded corners.
It's still not perfect but about as good as it gets on a springy bowden system like the Vyper has.

Jerk Values that I used:
X:8 Y:8 Z:0.4 E:12

[image]

rommulaner, 2024-12-26:
I have also tested it and saw better results with Classic Jerk.
I added the Classic Jerk variants in the current release.
Thank You!

closed 2024-12-26 by rommulaner

kb-elmo, 2025-01-01:
Very nice! Thank you 😄