rommulaner/Anycubic_Vyper_MB_CE_6.2 #53: Not work a BedLeveling!

PlasmaKirill, 2025-03-12:
Hello. I have a Vyper stock.
Set a LCDFirmware - [Vyper-CE-6.2.2-Touchscreen-2025-01-20.zip](https://github.com/rommulaner/Anycubic_Vyper_LCD_CE_6.2/releases/download/Vyper-CE-6.2.2/Vyper-CE-6.2.2-Touchscreen-2025-01-20.zip)
Thats OK.
Then I try to set a MBFirmware - [main_board_LA_CJ_IS_4x4_AllMetalHotend.zip](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/blob/main/builds/main_board_LA_CJ_IS_4x4_AllMetalHotend.zip)
Thats OK. Worked, But!....
Made a PID-Calibration. OK.
I cant made a Bed-Leveling - nozzle and table arу heating, than nozzle Just go to center of the table and stop - 30 min will waiting!
Sensor is work - red lamp is light when I press to a nozzle.
Please help! ))

rommulaner, 2025-03-12:
Hello!
Normally such a behavior occurs when the sensor got triggered too early. Make sure that the harness and the PTFE tube can move freely and the cable is fixed to the extruder. Otherwise it will move when gantry is moving towards the bed and triggers the sensor. 
Additionally make sure that the LCD OS has been flashed to V3.5. You can check that by using an microSD with an empty DWIN_SET folder on it in the screen and power on the printer.
Then you can see the version on the top of the blue screen.
Greets
Robert

PlasmaKirill, 2025-03-12:
yes - version is 3.5.
cable is OK too. 
Now I set a factory firmware 2.4.5 - she is worked perfect and leveling is OK, but many functions is locked, :(

PlasmaKirill, 2025-03-12:
> So you did PID and started to probe the Base Z-Offset and it stops there?
yes.
> Does the nozzle go down when it's at the center of the bed and stops after some mm?
yes. First nozzle stop on 1mm, then touch a cover and stop forever ))
> Can you measure the distance between the nozzle and the bed after homing and moving to Z=0?
now I have a factory firmware anв try to configure her.

rommulaner, 2025-03-12:
I think that the nozzle is too far away from the bed at your setup and that's why it stops. Adjusting the endstops or using a stock Hotend will solve that.

PlasmaKirill, 2025-03-12:
> I think that the nozzle is too far away from the bed at your setup and that's why it stops. Adjusting the endstops or using a stock Hotend will solve that.
In stock firmware calibration worked fine! I think its problem not in nozzle or else in mechanical! Its program problem!

rommulaner, 2025-03-12:
The Community Firmware has some different settings as the the stock firmware. Someone had the same issue with the bedmesh.
Herr is the Issue you can read and solve: https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2/issues/48