rommulaner/Anycubic_Vyper_MB_CE_6.2 #55: Issue with the "Print Test Pattern" function

Kenwood01, 2025-05-18:
With the “Print Test Pattern” function, the function is never completed correctly. It prints everything, but the print head does not leave the area. Where it should normally move away and make the print accessible. If I remember correctly, the display says “leaving G26” which it does not do. Canceling it does not help then it says “Canceling” but nothing happens. Only switching it off and on again solves the problem.
The second problem is more of a question. I first used the “main_board_LA_CJ_IS_4x4_FSD” firmware. There you can also set the individual values in “View Leveling Mesh”. However, with the firmware “main_board_LA_CJ_IS_7x7_FSD” it does not change values. Is this the normal behavior of the firmware?

rommulaner, 2025-05-18:
Hi!
The firmware only executes the existing G26 from Marlin which unfortunately can't be aborted. So it's directly related to Marlin and not the CE-Firmware.
With the 4x4 version you can edit all the probed and shown values, that's right. With the 7x7 it's not possible to show all values, so it makes no sense to edit only the shown values. That's why it's not possible with the 7x7.

Kenwood01, 2025-05-18:
Ok thanks for the answer and the information

closed 2025-05-18 by Kenwood01